EnhancedRadioDevices/HamShield #22: Cant compile packet examples, missing DDS.h, packet.h

Temtel, 2016-08-27:
Just got my two HS-09's.  Working great otherwise, but can't compile any of the AFSK & packet examples due to missing libraries.  I'd really like to experiment with packet; where can I get versions of these libraries that were meant to work with these examples?

Temtel, 2016-08-27:
Ahh, found them,  Looks like you have to install other libraries to use these examples.
https://github.com/EnhancedRadioDevices/DDS/tree/master/src
https://github.com/EnhancedRadioDevices/HamShield_KISS/tree/master/src

mogar, 2016-08-27:
Hi Temtel,

We're just finishing up revamping our libraries. Including all of the KISS and DDS stuff in the standard HamShield library caused even simple sketches to use a lot of memory.  We broke those out into separate libraries to make things a bit easier for people who didn't want to do packet, but we're still documenting how to use the libraries.

We plan to get the libraries integrated with Arduino's library manager soon, which will make this all quite a bit easier.

closed 2016-08-27 by mogar