EnhancedRadioDevices/HamShield #23: AX25Receive.ino unable to receive APRS Packets

Temtel, 2016-09-07:
Compiles fine, I can transmit AFSK successfully with other sketches, and I can hear the AFSK packets on 144.390Mhz through the audio jack, but so far have been unable to decode AFSK using any example sketch, including AX15Receive.ino.   There is plenty of APRS traffic on 144.390 in my area.  Shouldn't I be able to decode it? 
If I'm reading the schematic correctly, looks like AFOut from the 1846 chip gets amplified by the AD8591 amp, then coupled to ARD_AUDIO_OUT, which traces to A2 on the Arduino.  Where in the code, do we tell Arduino to listen on A2 for AFSK signals?

Temtel, 2016-09-08:
I can decode packets heard from a speaker attached to the hamshield, via microphone on APRSDroid, but still no decode in serial monitor.

mogar, 2016-09-17:
There's a bug with APRS receive. We're working to get it resolved, and we'll let you know as soon as we have a patch for it.

Temtel, 2016-10-13:
Morgan fixed this bug with some changes to the HamShield_KISS library on 9/24/2016.  If you obtain the latest library, AFSK Receive should work.

closed 2016-10-13 by Temtel