EnhancedRadioDevices/HamShield #25: AX25Receive example fails to decode many packets

jimcam, 2016-11-12:
The AX25Receive example does not process all packets.   Frequency set to 144390.   A local transmitter is generating packets.  A local digipeater repeats the packet.   These packets are clearly heard in the receive audio through the headphone.  Many short test packets are decoded sporadically and most (perhaps all) normal APRS packets are not decoded at all.

mogar, 2016-11-14:
@jimcam Can you give me more details about your setup? Specificaly, what are you using to generate the APRS packets and what digipeater are you using?

jimcam, 2016-11-14:
@mogar I am using multiple configurations.  All are working well on the APRS network but none of the packets generated here are processed by the HamShield. 

1) a digipeater (N7QME-1) -- APRX running on a RPi connected to a KPC-3 in KISS mode.   5 watt mobile to an external antenna.  APRS.fi is reporting good packets on this digi.

2) a kenwood TH-D72a HT (N7QME) to monitor APRS packets and to generate position reports to exercise the digi and to test the HamShield.   The digi picks up these packets and repeats without trouble.

3) a packet injector --  A script on the RPi creating valid APRS weather station packets sent to a KPC-3 with a normal serial interface.  The radio is an old HT running 1/2 watt and antenna just enough to have the digi receive and repeat the packets generated.   I use this to put a number of different weather stations on the air so that the HT can display the weather.    These packets go through the digipeater without problems.

jimcam, 2016-11-14:
@mogar At this point I've traced it to currentPacket->crcOK() returning false inside AFSK::Decoder::read()

jimcam, 2016-11-18:
Using a different HamShield and the Elegoo UNO R3, the APRS packets are decoded properly.

The failing setup is using an original Arduino UNO.  Any known issues?   I will test the HamShield from the failing setup with the UNO R3 tonight or tomorrow.

esteward, 2016-11-18:
@jimcam Thanks for the update. Surprising that it's working on one setup but not another. There aren't any known issues with the HamShield09s on any Arduino variant. Please keep us posted with the results of your R3 test.

It's possible that the failing arduino has a poor 16MHz oscillator, and so the phase information of the received signal isn't being decoded properly.

mogar, 2016-11-18:
@jimcam Thanks for the update. Surprising that it's working on one setup but not another. There aren't any known issues with the HamShield09s on any Arduino variant. Please keep us posted with the results of your R3 test.

It's possible that the failing arduino has a poor 16MHz oscillator, and so the phase information of the received signal isn't being decoded properly.

jimcam, 2016-11-19:
Twilight zone.  Nothing works at the home QTH in any combination.  Two sets don't even talk to each other! 
The success with the Elegoo UNO R3 and second shield was at a remote location.    Now I will take both units back to the remote location for testing on Monday.

BTW: The original Arduino UNO is only 250Hz low in frequency.  The R3 is 950Hz low.  Both should be within the capture range of discriminator.  Correct?    Also the service monitor is indicating only 3KHz deviation.   Is there a way to increase the deviation to better match the HT and digipeater levels?

jimcam, 2016-11-25:
Two separate problems:
1) The UNO R3 works with both HamShields to decode packets in an RF quiet environment.   It will not work at the home QTH but does at the remote site where the noise does not exist.
2) The original Arduino UNO board that I had in stock does not work to receive and decode packets even in the RF quiet environment.

mogar, 2016-11-27:
The Uno/R3 difference is probably an oscillator issue. The Uno R3 has a different oscillator than the original Uno, which means that the timers used to calculate phase information in the decoder may be more accurate. It should be possible to make the decoder more robust to clock noise. I'll look into that. 

The noise issue with the R3 is probably a separate issue, but the fix may be the same (making the decoder more robust).

This is a bit of a project, but I'll put it on our to-do list.

Dinsmoor, 2018-03-08:
Hello,

Has there been any development on this issue? I'm looking at purchasing this board for reasons as described in the OP, however these issues would prevent that if the bug still exists.

73

Dinsmoor, 2018-03-08:
@mogar Thanks for the update, it's much appreciated.

jimcam, 2018-03-09:
@mogar that’s good news on the audio chain improvements.  Looking forward to the results of your testing.   Is there any way to retrofit it back into my three original kickstarter units?
Jim.

joshbuhler, 2018-03-25:
I've got a Hamshield 9 connected to an Elegoo Uno R3. Sending packets using the AFSK_SerialMessenger sketch seems to work ok, but when I run the AX25Receive sketch tuned to 144.39, I'm not seeing any packets being decoded. I can hear APRS packets coming in through the audio, and on another radio, they're being decoded just fine.

Is the Hamshield 9 compatible with the Elegoo? I just haven't had any luck with any of the example sketches actually decoding, or showing any evidence of trying to receive & decode AX.25.

kghill, 2018-11-03:
Wondering what the status is regarding this issue? I have the new Hamshield hardware and the latest software upload. I am using the Geekcreit UNO R3. The data after compiling states the oscillatorboard is 28.800 khz. Is this acceptable.

mogar, 2018-11-08:
Thanks for your patience here. I just pushed a couple of changes to the HamShield library (this one) and the HamShield_KISS library. Those changes are all on the master branches, but they haven't been released to the Arduino library manager yet.

With these changes, I'm successfully decoding packets at 144.39MHz. I'm using a co-located transmitter, so my signal-noise ratio is pretty good.

Please try pulling down these changes and testing them out. We're going to do a bit more testing on our end, and will create a new Arduino release as soon as we're satisfied.

kghill, 2018-11-09:
I am happy to report that I am receiving packets on the Putty terminal emulator and on the APRSSIS32 reporting application. Not sure what the RSSI is. I am using a 1/4 wave antenna mount on my vehicle in my garage and receiving packets from a repeater 20km away.  I will be doing further testing. Like to figure out how I can read the RSSI.  

73 and Cheers, VE6AGD.

closed 2019-07-14 by mogar