EnhancedRadioDevices/HamShield #26: Compile Errors 'TIFR2' was not declared in this scope AFSK_SerialMessenger (and others)

Temtel, 2016-11-19:
Errors are failing compile of some sketches for several boards. Tested compiles for other boards as well, following is an example of the output:

<pre>
Arduino: 1.6.12 (Windows 7), Board: "Arduino Leonardo"

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:30:0,

                 from sketch\AFSK_SerialMessenger.ino.cpp:1:

C:\Users\JT\Documents\Arduino\libraries\HamShield-master\examples\AFSK_SerialMessenger\AFSK_SerialMessenger.ino: In function 'void TIMER2_OVF_vect()':

C:\Users\JT\Documents\Arduino\libraries\HamShield-master\examples\AFSK_SerialMessenger\AFSK_SerialMessenger.ino:141:5: warning: 'TIMER2_OVF_vect' appears to be a misspelled signal handler, missing __vector prefix [-Wmisspelled-isr]

 ISR(TIMER2_OVF_vect) {

     ^

AFSK_SerialMessenger:142: error: 'TIFR2' was not declared in this scope

   TIFR2 = _BV(TOV2);

   ^

In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:90,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28,

                 from sketch\AFSK_SerialMessenger.ino.cpp:1:

AFSK_SerialMessenger:142: error: 'TOV2' was not declared in this scope

   TIFR2 = _BV(TOV2);

               ^

exit status 1
'TIFR2' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences
</pre>

Results for compile of AFSK_SerialMessenger by board:
<pre>
Failed  Yun(TIFR2 was not declared in this scope)
Success Uno
Success Duemilanove or Decimelia
Success Nano
Success Mega or Mega 2560
Success Mega ADK
Failed Leonardo(TIFR2 was not declared in this scope)
Failed Micro(TIFR2 was not declared in this scope)
Failed Esplora (TIFR2 was not declared in this scope)
Success Mini
Success Ethernet
Success Fio
Success BT
Failed Lillypad USB (TIFR2 was not declared in this scope)
Success Lillypad
Success Pro or Pro Mini
Failed NG or older (Not enough memory)
Failed Robot Control (TIFR2 was not declared in this scope)
Failed Robot Motor (TIFR2 was not declared in this scope)
Failed Gemma (Serial was not declared)
</pre>

mogar, 2016-11-21:
Thanks for the exhaustive report, Temtel.

At the moment, the default HamShield library works with the Leonardo (and should work with the other variants whose error was TIFR2). However, a lot of the code that requires the DDS won't work yet. The DDS requires hardware timers, and different Arduino variants have different timers. We're working on adding support for the timers in the other variants, but that's a few weeks out at least.

skitzoid799, 2016-11-24:
That makes sense-- thank you.

jimcam, 2017-01-31:
@mogar  Thanks for working on supporting DDS on YUN and Leonardo.  I am hoping to use my YUN soon!

mogar, 2017-03-01:
We've decided not to officially support the Yun or Leonardo because the timers required interfere with millis() and delay(). Either we mess up the time base of the arduino, or we can't hit the frequencies needed for AFSK with the pin 3 PWM.

If you'd like to do the work yourself (and possibly mess up timing on the arduino for other functions), then porting the DDS and AFSK code to the arduino is mostly an exercise in swapping all the Timer 2 registers to Timer 0 registers in DDS.c and packet.c.

closed 2017-03-01 by mogar