EnhancedRadioDevices/HamShield #27: Does the KISS example work with YAAC?

jimcam, 2016-11-28:
I'm using the HamShield with the R3 UNO board connected to my mac running the latest JAVA and YAAC.    I have configured the radio interface to connect to the board but see no APRS packets coming through on the RAW Packet Sniffer window.   Has the KISS mode example been used with YAAC?

mogar, 2016-12-21:
@jimcam The KISS script hasn't been tested with YAAC, and I'm guessing that it probably won't work by default. We generally just log the KISS output straight to a text file with PUTTY or something similar.

It shouldn't be too hard to add the necessary syntax/commands to the KISS example to make it work with YAAC. Once that's done, configure YAAC: you'd want to set it up to use serial-KISS TNC port and then set the COM port correctly and the baud rate to whatever is in your script (it's 9600 in the KISS script by default). 

I'll put this on our to-do list for a new example.

mogar, 2016-12-27:
@jimcam I got the KISS example talking to YAAC today. There were a few minor issues in that example that was causing some problems. I've pushed my changes to master, so you can take a look at those now if you want. You'll have to check out the latest HamShield and HamShield_KISS masters.

If you want to wait for an official, tested release, that should be happening in the next week or two.

mogar, 2016-12-27:
Oh, also: configuring YAAC to use the HamShield KISS TNC is done as follows:

1. open the configure YAAC wizard
2. follow the prompts and enter in your details until you get to the "Add and Configure Interfaces" window
3. Choose "Add Serial KISS TNC Port"
4. Choose the COM port for your Arduino
5. set baud rate to 9600 (default
6. set it to KISS-only: with no command to enter KISS mode (just leave the box empty)
7. Use APRS protocol (default)
8. hit the next button and follow directions to finish configuration

You should then be able to view all packets received from the "view->Raw Packets" window. I used the AFSK_SerialMessenger example to send test packets.

closed 2016-12-27 by mogar

jimcam, 2017-01-31:
@mogar   Thanks for the KISS fix.   Now receiving APRS packets into YAAC.