EnhancedRadioDevices/HamShield #29: KISS mode crashes when transmitting back to back packets leaving the transmitter on

jimcam, 2017-01-31:
While testing messaging from YAAC and the latest KISS code, I noticed that isolated TX packets are less likely to be received by my local digi and by my HT when sent as an isolated packet however if the message send button is pressed a second time quickly after the first packet is sent then the message is received by both the digi and the HT.   If sent too quickly or too many (not sure which) the Arduino locks up leaving the transmitter on.

mogar, 2017-02-03:
I expect this is a bug in the KISS.loop() function in how it handles commands when it's in the middle of executing a different command. The best way to handle this may be to put a check in the loop function before line 26 of KISS.cpp to make sure it doesn't start parsing serial data when it's in the middle of some other task.

mogar, 2017-02-03:
I added a check in the loop function. Could you please pull the waitWhileBusy branch of HamShield_KISS and give it a try?

jimcam, 2017-02-03:
@mogar Works great!  Thanks for the quick response.

closed 2017-02-03 by jimcam