EnhancedRadioDevices/HamShield #30: KISS mode tx packets are not being transmitted cleanly.

jimcam, 2017-02-01:
While testing messaging from YAAC and the latest KISS code, I noticed that isolated TX packets are less likely to be received by my local digi and by my HT when sent as an isolated packet however if the message send button is pressed a second time quickly after the first packet is sent then the message is received by both the digi and the HT.    Sounds like the modulation starts weak and builds and by the time the second packet starts the entire message is receivable.

mogar, 2017-02-03:
@jimcam Your analysis seems correct. You can even hear it if you you're doing phone transmissions. When you key up, the audio comes through very weak, but it gets stronger after about 5s. 

The gain control tables for the AU1846 isn't optimized right now, and could use some tweaking. Those values are set when the HamShield is initialized (see lines 362 to 400 in HamShield.cpp in the library).

nigelvh, 2018-02-14:
@jimcam I believe this issue has been fixed in more recent revisions of the library. Testing now I don't get the initial weak audio. Please re-open if it remains an issue for you.

closed 2018-02-14 by nigelvh