EnhancedRadioDevices/HamShield #31: What is the correct config for the KISS mode interface to APRX?

jimcam, 2017-02-03:
My HamShield loaded with the latest KISS SW works for RX with YAAC.  Now I'm trying to interface the HamShield to APRX to get RX packets into APRX to use it as an RX only iGate.    Here is the interface section of my aprx.conf.

<interface>
   serial-device /dev/ttyACM0  9600 8n1    KISS
</interface>

In the aprx.log file see this...
2017-02-03 04:57:30.218 TTY /dev/ttyACM0 opened
...but no RF packets logged to aprs-rf.log

Any ideas?
Thanks,
Jim.

mogar, 2017-02-03:
Are you sure /dev/ttyACM0 is the right port for the Arduino? I would expect it to be something like /dev/ttyUSB0 or something similar. Can you verify that port? If you just set up a serial console to that port, do you see any packets coming through there?

jimcam, 2017-02-03:
/dev/ttyACM0 is the device that gets added to /dev when I plug in the Arduino.

I just had it plugged into my laptop for the waitWhileBusy testing so I moved it back to the RPi.   APRX polls the device.  Here is the log:

2017-02-03 23:35:29.535 TTY /dev/ttyACM0 failed to open; errno=2 (No such file or directory)
2017-02-03 23:36:00.437 TTY /dev/ttyACM0 failed to open; errno=2 (No such file or directory)
2017-02-03 23:36:35.441 TTY /dev/ttyACM0 opened

I have not tried setting up a serial console.   I will go (re)learn how to do that and give it a try.

jimcam, 2017-02-04:
After killing aprx I use this command:  
sudo screen /dev/ttyACM0 9600 8N1
to connect to the Arduino.   Nothing comes through.

Using this command while using screen or aprx: 
sudo stty -F /dev/ttyACM0
I can verify that the baud rate is set correctly.

Next step will be to load an echo program into the Arduino to verify that the Arduino can talk to the Pi.

jimcam, 2017-02-04:
Echo program works on the RPi using screen to talk to the USB Arduino.   Mystery as to why the KISS load does not work on the RPi.   What to try next?

jimcam, 2017-02-04:
Built serial echo into KISS at the outer loop.   It echoes.   Packet decode appears to be not working.  Perhaps there is local noise in the area of the radios or from the RPi itself.   Let's test that...

Further investigation:
Earlier I had found that packet decode was more reliable with the squelch off so by plugging in headphones I should here an open squelch.   With the Arduino plugged into the Mac laptop the LED on the HamShield flashes a few times then comes on solid and the squelch is open with noise in the headphone.   With it plugged into the RPi the lights do exactly the same flashes but there is no audio noise at all.   Using the Arduino reset button to restart the KISS program -- no change.    

Why audio on one platform but not the other?   I think I'm out of things to try.   Any ideas?

jimcam, 2017-02-04:
Aha!   A noisy power problem.  Everything appeared to be running correctly on the USB power but with external power connected then the RX started working.

closed 2017-02-04 by jimcam