EnhancedRadioDevices/HamShield #32: Support for Intel Edison

jimcam, 2017-02-06:
Attempting to build an APRS iGate.   (Wireless) Internet connectivity is essential.   Intel Edison on the Arduino carrier board makes a compelling package.

This configuration may also support an IRPL node.   This needs investigation.

mogar, 2017-02-07:
Fully agreed. Edison support is probably behind Leonardo in priority, but we do intend to do that.

jimcam, 2017-06-04:
I'm ready to integrate the HamShield with the Intel Edison Arduino.   I have all the parts but am missing the foundation SW.   I would like to get BLE sensor data sent as telemetry data over APRS.   I'm pulling together the BLE listener pieces on the Edison but I still missing the DDS timer support to run APRS.

mogar, 2018-11-08:
Closing this since Intel Edison is no longer available.

Long term, we do want to break out the timing functions to make HamShield more extensible for other architectures.

closed 2018-11-08 by mogar