EnhancedRadioDevices/HamShield #38: Unclear documentation for DTMF decoding/encoding

Chris2fourlaw, 2018-05-30:
I can't seem to find documentation for DTMF usage other than scattered references in the wiki.

mogar, 2018-06-05:
We are working on a DTMF example sketch to show how it's used.

Chris2fourlaw, 2018-06-05:
Sounds good

marcushammar, 2018-06-05:
Thanks! Please include an example on how to receive DTMF. I plan to control our gingerbread house next Christmas with an amateur radio and DTMF tones.

mogar, 2018-06-25:
I've pushed a few commits to master that show to use DTMF. I ended up adding a few functions to the library to make things easier.

I also added a DTMF example that shows TX and RX. 

We'll be releasing a new version of the library to Arduino after we make some tweaks to other functionality.

closed 2018-06-25 by mogar