EnhancedRadioDevices/HamShield #41: DTMF sketch, codes transmitted seperately, sounds jumbled.

Chris2fourlaw, 2018-07-01:


mogar, 2018-07-03:
I adapted the DTMF example to send all DTMF tones in the buffer before turning off TX. Take a look and let me know if it works for you.

Chris2fourlaw, 2018-07-03:
Can you add 20ms before the cutoff? last number gets partially cut off. Other than that, works great!

Chris2fourlaw, 2018-07-03:
also invalid codes break the transmission

mogar, 2018-07-03:
Ok, I've updated for both of those issues. LMK what you think.

Chris2fourlaw, 2018-07-03:
Great. Sorry to be so picky, but 20ms at the beginning would be good too. You have to time it in the loop so that the first one doesn't get cut off. Also, there's a weird noise at the end of each code. Sound file attached; slightly hard to hear but its apparent on the radio. I'll try inspecting with it to see if it affects HT decoding.

Chris2fourlaw, 2018-07-04:
HT won't decode because the codes aren't sent fast enough 😞

mogar, 2018-08-05:
I've updated the library with some functions that support manual timing for DTMF. Please take a look at the DTMF example in master and let me know how it works for you.

closed 2018-11-08 by mogar