EnhancedRadioDevices/HamShield #42: DTMF sketch, doesn't detect codes

Chris2fourlaw, 2018-07-02:
Transmitted 25W, headphone jack sounded loud and clear, but monitor didn't show right codes.

marcushammar, 2018-07-02:
I have also been having issues to decode DTMF. Tried the new code last weekend and it was not working for me properly. Sending DTMF was working with the new code thou.

mogar, 2018-07-03:
Hmm. What are you using for your transmitter?

You may want to try turning up Volume1 and seeing if that helps.

Chris2fourlaw, 2018-07-03:
Raising volume 1 creates lots of false positives and no correct reads

mogar, 2018-07-03:
Is your SQ set correctly?

What are you using for your transmitter? Can it accurately receive DTMF codes from the HamShield?

Chris2fourlaw, 2018-07-03:
Using SQ in the sketch. 25W is a old kenwood mobile. HT is 5W and Baofeng UV-5X3. Trying receiving/decoding with the HT now.

mogar, 2018-11-08:
Did you end up getting this working? We've pushed some updates to the DTMF code, but if it's still not working for you I can try to dive a bit deeper on the issue.

Chris2fourlaw, 2018-11-22:
I haven't tested it yet. I should be able to soon

mogar, 2019-07-14:
Recent updates seem to be working. Closing this for now. Please re-open if it's still not working for you.

closed 2019-07-14 by mogar

AllDigital33, 2019-11-15:
Mogar, I've got the same issue with two separate HamShields. I bought the HamShield Mini last Summer connected to an Arduino Due and I had a lot of issues with it receiving and recognizing DTMF tones. It would recognize 1 and 3 and sometimes 6 or 7, but not reliably. It would send DTMF just fine. I then ordered the regular HamShield and paired it with a new Arduino Uno. Using the same example code I get the exact same behavior. I am trying it with a Baofeng UV-5R5 and with a Yaesu VX-8. Both handhelds send great DTMF, sound clean, and trigger local repeaters, so I don't think it is the radios. The issues sound very similar to the issues above. I have tried increasing the volume1, adjusting the squelch, DTMF detect times, etc. but nothing seems to work. I've also swapped out a few different antennas. Please advise.