EnhancedRadioDevices/HamShield #44: KISS example compile error: call of overloaded 'frequency(long int)' is ambiguous

kghill, 2018-10-24:
I have download the latest Hamshield examples and libraries (DDS & Hamshiled_KISS and when I compile the KISS.ino example I get the following error: **call of overloaded 'frequency(long int)' is ambiguous.**

Any help would be much appreciated.

**Here is the error message summary.**

C:\Users\keng\Documents\0 Arduino\HamShield-master\examples\KISS\KISS.ino: In function 'void setup()':

KISS:59:25: error: call of overloaded 'frequency(long int)' is ambiguous

   radio.frequency(144390);

                         ^

In file included from C:\Users\keng\Documents\0 Arduino\HamShield-master\examples\KISS\KISS.ino:22:0:

C:\Users\keng\Documents\0 Arduino\libraries\Hamshield/HamShield.h:247:8: note: candidate: bool HamShield::frequency(uint32_t)

   bool frequency(uint32_t freq_khz);

        ^

C:\Users\keng\Documents\0 Arduino\libraries\Hamshield/HamShield.h:248:14: note: candidate: bool HamShield::frequency(float)

         bool frequency(float freq_khz);

              ^

Using library Hamshield in folder: C:\Users\keng\Documents\0 Arduino\libraries\Hamshield (legacy)
Using library HamShield_KISS-master at version 1.0.4 in folder: C:\Users\keng\Documents\0 Arduino\libraries\HamShield_KISS-master 
Using library DDS-master at version 1.0.3 in folder: C:\Users\keng\Documents\0 Arduino\libraries\DDS-master 
exit status 1
call of overloaded 'frequency(long int)' is ambiguous

kghill, 2018-10-25:
I was able to compile the KISS example by hacking around the issue. I set the default frequency to 144390 in HamShield.cpp on line 244 and commented the radio.frequency(144390); on line 59. 

Hope this helps.

kghill, 2018-10-27:
I am having a similar problem with Serial Transceiver example. The following is the compile error I am getting.

Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\keng\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\keng\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\keng\Documents\0 Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10804 -build-path C:\Users\keng\AppData\Local\Temp\arduino_build_435968 -warnings=none -build-cache C:\Users\keng\AppData\Local\Temp\arduino_cache_236843 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.avrdude.path=C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14 -prefs=runtime.tools.avr-gcc.path=C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -verbose C:\Users\keng\Documents\0 Arduino\0 HamShield-master\examples\SerialTransceiver\SerialTransceiver.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\keng\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\keng\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\keng\Documents\0 Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10804 -build-path C:\Users\keng\AppData\Local\Temp\arduino_build_435968 -warnings=none -build-cache C:\Users\keng\AppData\Local\Temp\arduino_cache_236843 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.avrdude.path=C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14 -prefs=runtime.tools.avr-gcc.path=C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -verbose C:\Users\keng\Documents\0 Arduino\0 HamShield-master\examples\SerialTransceiver\SerialTransceiver.ino
Using board 'uno' from platform in folder: C:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23
Using core 'arduino' from platform in folder: C:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23
Detecting libraries used...
"C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino" "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\variants\standard" "C:\Users\keng\AppData\Local\Temp\arduino_build_435968\sketch\SerialTransceiver.ino.cpp" -o "nul"
"C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino" "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\variants\standard" "-IC:\Users\keng\Documents\0 Arduino\libraries\Hamshield" "C:\Users\keng\AppData\Local\Temp\arduino_build_435968\sketch\SerialTransceiver.ino.cpp" -o "nul"
"C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino" "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\variants\standard" "-IC:\Users\keng\Documents\0 Arduino\libraries\Hamshield" "C:\Users\keng\Documents\0 Arduino\libraries\Hamshield\HamShield.cpp" -o "nul"
"C:\Users\keng\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino" "-IC:\Users\keng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\variants\standard" "-IC:\Users\keng\Documents\0 Arduino\libraries\Hamshield" "C:\Users\keng\Documents\0 Arduino\libraries\Hamshield\HamShield_comms.cpp" -o "nul"
[... cut ...]
C:\Users\keng\Documents\0 Arduino\libraries\Hamshield/HamShield.h:247:8: note: candidate: bool HamShield::frequency(uint32_t)

   bool frequency(uint32_t freq_khz);

        ^

C:\Users\keng\Documents\0 Arduino\libraries\Hamshield/HamShield.h:248:14: note: candidate: bool HamShield::frequency(float)

         bool frequency(float freq_khz);

              ^

Using library Hamshield in folder: C:\Users\keng\Documents\0 Arduino\libraries\Hamshield (legacy)
exit status 1
call of overloaded 'frequency(long int&)' is ambiguous

mogar, 2018-10-30:
Hey folks, sorry for the issue. We've update the library to fix this. Please see the v1.1.1 version of the library and let us know if you have any other issues.

closed 2018-10-30 by mogar

kghill, 2018-11-03:
It compiles. Thanks for the update.