EnhancedRadioDevices/HamShield #45: KISS Example Decodes Packets, AFSK Messenger Doesen't

aaron9412, 2019-02-03:
I have a test environment set up with a baofeng UV-5R radio connected to my android phone running APRSdroid which I can use to send messages through the hamshield and an arduino uno to my PC.

If I use the KISS example, I can receive messages and display them on a serial terminal (garbled, as expected) or cleanly in YAAC. I am able to repeat this and it works almost all the time, minus a few missed or rejected packets.

However: if I upload the serial messenger example, no packets are ever displayed on the serial terminal. Just a pretty, blank, white screen. Yes, I made sure the port was not being used by something else, yes, I made sure the frequency was set to the frequency I'm using. There are just no packets ever reported to the serial terminal. 

Any Ideas?

-Aaron

mogar, 2019-02-03:
The serial messenger example was made as a way to easily send packets back and forth between HamShields. I think the packet structure is different than APRS packets. If you have two HamShields, you could try using SerialMessenger with both and seeing if that works for you. Let me know if that doesn't work, as it means something else is broken.

If that does work for you and you want a SerialMessenger sketch to receive APRS packets, we can point you in the right direction for that.

aaron9412, 2019-02-04:
Please point me in the right direction, that's exactly what I'm looking for!

The AFSK_SerialMessenger says that it sends messages over APRS in the intro text, that's why I got confused. 

> /* Hamshield
> ** Example: AFSK Serial Messenger
> ** Serial glue to send messages over APRS.

mogar, 2019-02-10:
You only have to make one or two changes to get it working with APRS:

1. make sure the frequency is set correctly on line 55
2. comment out the setSQOn function call on line 61
3. uncomment the bypassPreDeEmph() function call on line 62

You're right that the description in the header is confusing. Next week I'll update the sketch with these changes so it matches the explanation.

Thanks,
Morgan

aaron9412, 2019-02-10:
Welllll.... I feel stupid. I've tried it like 8000 different ways, I guess I just didn't hit the right combination.

Receiving data works great!:
[image]

I can receive packets quite reliably. 
I've only been able to send one packet back to my phone (been playing with it for the past hour).
Any suggestions, I'm all ears, but it may be my setup or too much noise in the environment for the hamshield to transmit over. Since I sent 1 packet I at least know it possible now.

Thank you, and sorry for wasting your time over something so simple!
-Aaron

closed 2019-02-10 by aaron9412

mogar, 2019-02-11:
Glad you got it working! Let us know if there's anything else you need.

As for transmitting, that's a lot harder to help with without having a similar receiver to test. One thing you might try is adjusting tx power or playing around with VoiceHpf and VoiceLpf functions.

aaron9412, 2019-02-26:
Thank you! I will try both.