EnhancedRadioDevices/HamShield #46: SpeechTX does't send the audio.

robson-sampaio, 2019-07-09:
I've just go my Hamshield and I have followed the steps to working on it, but I am having some issues when I try to transmit the signal to my motorola talkabout. I conected the hamshield to my arduino uno, plug it on a wall power supply, plug it to the arduino cable and uploaded the SpeechTX code to it. I had to change de the frequency in the code to 462562 in both Setup and Loop functions but nothing happens. Although I can comunicate from my motorola radio to the hamshield, but the other way doesn't work. Any clue??

mogar, 2019-07-17:
I just tested it out and it seems to be working for me. There are a couple of things you can try:

1. Make sure you're using an Arduino that has an ATMega168 or ATMega328 at 16MHz. That's required for the talkie library that we're using.
2. Make sure you have the talkie library installed. You should now be able to do this from the Arduino's library manager.
3. Make sure you're listening from the start of the sketch. It starts talking immediately, but after it finishes it doesn't keep talking.

Let me know if that works for you.

robson-sampaio, 2019-07-29:
I just had to use danger mode to operate on the frequency of 462MHz. Now it is working fine. Thank you!

closed 2019-07-29 by robson-sampaio