EnhancedRadioDevices/HamShield #48: Hamshield mini packet transmission size

gbaranedavid, 2020-12-30:
Hello team, I am currently working on a project using the Hamshield mini. 

I am exploring the AFSK_SerialMessenger and I noticed a header indicating that I can transmit a message under 254 characters so I have a few questions and they are as follows:

1. Does the hamshield mini have the capacity to transmit over 10,000 characters at time.
2. If yes, what and what am I likely to adjust the AFSK_SerialMessenger codes and on the radio(if any) to enable me achieve a fluent transmission of 10,000 text characters or above.

3. If no, please can you advice me on my other options.

Thanks

mogar, 2021-04-19:
The AFSK_SerialMessenger app was originally built to support APRS, which has a maximum packet size of 256 bytes. We'll be leaving the official library as is for this reason.

It should be possible to update the code to support larger packet sizes, but there will be a lot of bookkeeping variables to changes. In particular, there are a variety of counter variables (like currentByte, readPos, dataPos) in packet.h that are defined as uint8s. Those would need to be changed to larger variable sizes. I haven't tested this, so you may run into some other issues as well.

The main limitation probably has less to do with the HamShield library than with hardware. I'm not sure you'll be able to hit 10k chars in a packet if you're running this on an Arduino. The Arduino Uno only has 2k of memory in the first place. If you used a Raspberry Pi, you shouldn't be hardware constrained. For details on that, see here: https://github.com/EnhancedRadioDevices/HamShieldPy (you'll still have to modify the library to get this to use higher packet sizes)

closed 2021-04-19 by mogar

gbaranedavid, 2021-06-14:
We observed that the github repository link  you sent for the
library/source code is for hamshield mini; we just recently acquired your
hamshield LoRa edition because of  the features it posseses that your
hamshield mini doesn't; we would like to confirm if the reply you earlier
sent is applicable to your Lora edition. We also need the github repository
link for the Hanshield LoRa library/source codes

gbaranedavid, 2021-06-23:
Also, just recently, we got the raspberry pi zero w integrated with our hamshield mini,'
Pulled the source codes and examples on python for raspberry pi including all requisite configuration but we noticed that you don't have an implementation of your AFSK_SerialMessenger in the python implementation as is in arduino C language implementation; the only examples available in your hamshieldpy library are below:

AppSerialController.py
ArduinoLikeExample.py
CTCSS.py
DTMF.py
FMBeacon.py
FoxHunt.py
HandyTalkie.py
Morse.py
SSTV.py

Is there an equivalent implementation of your AFSK_SerialMessenger in python for pi zero w? 

If yes or no, please point us in the right direction. 

Thanks you very much